sandeepmistry/arduino-CAN #2: ESP32 - how can i setup another baud rate?

latonita, 2018-04-04:
HI,

After atmel +mcp2515 i'm trying to use ESP32 for can bus communication.
However I can't set custom speed. I need 83.333 kbps.
How did you calculated BTR register values? tried kvasir bit calculator, but can't get same numbers. dont know clock frequency at least.
I managed to use excel :) and to interpolate required bit rate with nice formula:
 BTR0 = 2989,2 * BAUD ^ -1,05
for 83.333 i got BTR0 = 29. And it works!!!! 👍 

```
    case (long)80E3:
      modifyRegister(REG_BTR1, 0x0f, 0x0c);
      modifyRegister(REG_BTR0, 0x3f, 30);
      break;
```

sandeepmistry, 2018-04-29:
@latonita it's based on https://github.com/nhatuan84/arduino-esp32-can-demo/blob/master/src/CAN.c#L204-L205

latonita, 2018-04-30:
Clear, thank you!

  
  

-- Sent from Yandex.Mail for mobile: http://m.ya.ru/ymail

closed 2018-08-12 by sandeepmistry