sandeepmistry/arduino-CAN #6: Doesn't work on SN65HVD230 Transceiver

ruffle-b, 2018-07-27:
Nice library but it doesn't work on a SN65HVD230 Transceiver (which are now very common).

Using your simple EngineRPM demo; it hangs at CAN.endPacket() (the standard can library in esp-idf does work with SN65HVD230).

Weirdly, if I put a Serial.println immediately before that it not only hangs but I have to power cycle my ESP32 before it will talk again:

```
CAN.write(0x0c); // engine RPM
Serial.println("Written 0x0c");
CAN.endPacket();
```

rahmaevao, 2018-08-16:
Hi, all.
It is problem have a solution.
Very important chip imaging.
Set the terminator to 60 ohms.
If the free standing on the bus is not 1.65 V, then use pull-up resistors for this.
CANH > RES 3kOhm > GND (Pull down)
CANL > RES 3kOhm > +3,3 V (Pull up)

rahmaevao, 2018-08-16:
Does not work with ADM3053 in standard piping

rahmaevao, 2018-08-16:
With braces also does not work

rahmaevao, 2018-08-16:
Why?

rahmaevao, 2018-08-17:
Driver ADM3053 also works.))

quadcodes, 2018-08-17:
I tried using with my Suzuki. I'm getting following data on CAN BUS.

```
Received packet with id 0x13 and length 4

?0p

Received packet with id 0x39 and length 0

Received packet with id 0x41 and length 0

```
How do I read/parse this data as OBD response ?

rahmaevao, 2018-08-17:
[image]

Your data is output if the LENGTH is not zero.

sandeepmistry, 2018-09-04:
@rahmaevao thanks for the tips!

sandeepmistry, 2018-11-03:
Closing due to lack of feedback.

closed 2018-11-03 by sandeepmistry