sandeepmistry/arduino-CAN #10: esp32 crashes on CAN.endPacket()

creativtransfer, 2018-11-18:
I have no transceiver attached and when sending a packet, the ESP32 hangs upon executing  CAN.endPacket();

Also when configured for loopback mode.

Which ESP32 Version do you use?

stiebel, 2018-11-28:
I have the same problem.
with no receiver CAN.endPacket() should return an error but hangs instead.
Same problem when sending RTR - hangs instead of returning error.
If not to fix I at least need a workaround. 
I work with a couple of WROOM32

noiasca, 2018-12-21:
Same here. Without transceiver ESP32 hangs in the 
while ((readRegister(REG_SR) & 0x08) != 0x08) {
and doesn't get beyond row 226

noiasca, 2019-01-08:
to get an error and avoid hanging of the lib i found a workaround and added an errorhandling based on my returncodes.

extend the while loop in file ESP32JA1000.cpp row 219:

 // wait for TX complete
  while ((readRegister(REG_SR) & 0x08) != 0x08) { // 0000‭1000‬ Transmission complete status. 0 = incomplete
    //Serial.print(F("D220 REG_SR:")); Serial.print(readRegister(REG_SR)); Serial.print(F("   REG_ECC:")); Serial.println(readRegister(REG_ECC));
    if (readRegister(REG_ECC) == 0xd9) {   // 217=‭11011001‬
      modifyRegister(REG_CMR, 0x1f, 0x02); // error, abort
      return 0;
    }
    if (readRegister(REG_SR) == 0x30) {   // e.g. if no transceiver is connected REG_SR=48 (‭0b00110000‬) ECC=0
      modifyRegister(REG_CMR, 0x1f, 0x02); // error, abort
      Serial.print(F("D227 abort ")); Serial.println(readRegister(REG_SR));
      return 0;
    }
    yield();
  }

Waged, 2020-04-08:
I got the same problem when using 500E speed and the solution for that is the 120 Ohm resistor , when I installed it, I find the error gone.

NeliusNDL, 2020-11-01:
> to get an error and avoid hanging of the lib i found a workaround and added an errorhandling based on my returncodes.
>
> extend the while loop in file ESP32JA1000.cpp row 219:
>
>  // wait for TX [...]

Thanks for this. I'll give it a try as I have the same problem. Is this still the best way to do it? Any updates to the libraries? Any better CAN bus library?

noiasca, 2020-11-01:
I don't see a change in the lib, so i guess this workaround still works.

DNThomas, 2020-11-10:
I have a transceiver attached (SN65HVD230). When plugged into a vehicle CAN bus, the lib performs as expected. When unplugged from the vehicle , it still hangs at the CAN.endPacket(); even with the above workaround.

I ended up just putting a timeout in place instead and that seems to work.

schnitzelPommesGMBH, 2021-03-16:
I have the same error but Im not sure if the transceiver is connected. it should but I can't verify it.

NeliusNDL, 2021-04-14:
look at this for a solution
https://github.com/sandeepmistry/arduino-CAN/issues/60#issuecomment-786664485

zengfanfan, 2024-04-10:
os this lib is out of maintenance ?

timurrrr, 2025-07-20:
Cross-referencing PR #160 for visibility as a potential fix

@sandeepmistry FYI