sandeepmistry/arduino-CAN #13: No communication two ESP32 with SN65HDV230 module

ogggi, 2019-01-12:
Hello,

I have two esp32 (DOIT devkit esp32) and two module with SN65HDV230. I used yours library. On one is load program CANsender and on second esp32 is program CANReceiver. First ESP32 sending me on serial message "sending packet...done" . My configuration is same like yours. I dont see on second ESP32 no messages. I testing change speed and pins and no reaction. 
I have more modules(ESP32 and SN65HDV230) and all of it is the same. 
Please help me.

Thank you

gerardwr, 2019-01-20:
Did you use a 120 Ohm resistor between CanH and CanL?

ogggi, 2019-01-20:
120 Ohm resistor is on board with SN65HDV230

gerardwr, 2019-01-20:
Was just a suggestion.

I use MPCP2551 and I had similar problems without the 120 Ohm.

Maybe you could try connecting the 2 Esp’s directly, so without the Transceivers.

 Connect GPio 4 and Gpio 5 of the 2 Esps crosswise. If that works with Cansender and Canreceiver then you are sure that the problem is in the Transceiver setup. If it does not work, the problem is in the HW or SW of the Esp32.

ogggi, 2019-01-27:
so, boards with SN65HDV230 from aliexpress are bad. Esp32 perfectly comunicate.

thank you for help

closed 2019-01-27 by ogggi

gerardwr, 2019-01-27:
Nice that it works now.

My MCP2551 transceivers work OK, but they need 5V power.

I'm waiting for some ordered SN65HDV230 because they need 3.3V power, just like the ESP32.

ogggi, 2019-01-27:
I bought three pieces and all are bad. On esp32 forum people writed that is bad series chips SN65HDV230 from china.

gerardwr, 2019-01-27:
That’s bad luck.

Let’s see how mine are when they arrive.

noripyon, 2022-02-09:
Old thread but FYI
MCP2551/2561 has 5V CMOS level interface with MCU and it violate VIH(min).

If you use 3.3VIO MCU, just use MCP2562 which has separate I/O supply pin.