sandeepmistry/arduino-CAN #14: Can Bus Filter ID´s and Formating Messages.

Petros144, 2019-01-18:
Thanks sandeepmistry for the Library!

I need a litte help with my Code.

I am working with a ESP32 and a TI SN65HVD230. Everything works fine and I can Monitor RPM and Speed from the Can Bus on my Smart fortwo Car.

my next goal is to show up Gears, I know that the transmission ECU is ID 0x418 and I also know how the message is set up.

for example: 

"ID = 0x418 Message = 4E FE 32 25 00 4E FE 37" This is for Neutral

"ID = 0x418 Message ="52 FE 30 68 BB 52 FE 30" This is for Reverse

I only tried a Filter with ID: =0x148

And Print the Bus to Console, what I Got is only 4 Bytes messanges and I need a 32 Byte variable to compare / mask. 

```
(if ID == 418 && Messsage == 4E FE 32 25 00 4E FE 37)
{
Serial.print("Neutral");
}
```

This is the Console Log:

```
------------------------------------
30
69
0
4E
FE
31
6A
0
4E
FE
35
6B
0
4E
FE
30
6C
0
.
.
.
-------------------------------------
```

there is so little information about the ESP32 Can Controller out there, so how to work this out.

maybe someone can help me out

gerardwr, 2019-01-20:
Maybe if you show the code we can help.

Petros144, 2019-01-20:
Here is My code that gives me only one Hex value at a time:

Its from the examples folder.

`
#include <CAN.h>

const bool useStandardAddressing = true;

void setup() {
  Serial.begin(9600);

  Serial.println("CAN OBD-II engine RPM");

  // start the CAN bus at 500 kbps
  if (!CAN.begin(500E3)) {
    Serial.println("Starting CAN failed!");
    while (1);
  }

  // add filter to only receive the CAN bus ID's we care about
  if (useStandardAddressing) {
    CAN.filter(0x418); // Adress of the Transmission ECU
  } else {
    CAN.filterExtended(0x18daf110);
  }
}

void loop() {
  if (useStandardAddressing) {
    CAN.beginPacket(0x7df, 8);
  } else {
    CAN.beginExtendedPacket(0x18db33f1, 8);
  }
  CAN.write(0x02); // number of additional bytes
  CAN.write(0x01); // show current data
  CAN.endPacket();

  CAN.parsePacket();
  Serial.println(CAN.read(),HEX); // Print out the Hex Message 

}`

gerardwr, 2019-01-20:
After CAN.parsePacket, CAN.read reads just 1 byte from the packet, and prints a new lines. This explains your log output.

The CANReceiver example shows how you can read the entire Packet. The while loop reads bytes from the package untill all bytes are read.

A good start is replacing the last 2 statements in your code by all statements in the example contained in “loop”.

closed 2019-02-25 by sandeepmistry

aladinbrahem, 2020-07-26:
i'm using this code after change frequency to 8e6
but in serial moniteur i still have just this line CAN OBD-II engine RPM with no ather error 
can you help me ?

#include <CAN.h>

// Most cars support 11-bit adddress, others (like Honda),
// require 29-bit (extended) addressing, set the next line
// to true to use extended addressing
const bool useStandardAddressing = true;

void setup() {
Serial.begin(9600);

Serial.println("CAN OBD-II engine RPM");

// start the CAN bus at 500 kbps
if (!CAN.begin(500E3)) {
Serial.println("Starting CAN failed!");
while (1);
}

// add filter to only receive the CAN bus ID's we care about
if (useStandardAddressing) {
CAN.filter(0x7e8);
} else {
CAN.filterExtended(0x18daf110);
}
}

void loop() {
if (useStandardAddressing) {
CAN.beginPacket(0x7df, 8);
} else {
CAN.beginExtendedPacket(0x18db33f1, 8);
}
CAN.write(0x02); // number of additional bytes
CAN.write(0x01); // show current data
CAN.write(0x0c); // engine RPM
CAN.endPacket();

// wait for response
while (CAN.parsePacket() == 0 ||
CAN.read() < 3 || // correct length
CAN.read() != 0x41 || // correct mode
CAN.read() != 0x0c); // correct PID

float rpm = ((CAN.read() * 256.0) + CAN.read()) / 4.0;

Serial.print("Engine RPM = ");
Serial.println(rpm);

delay(1000);
}