sandeepmistry/arduino-CAN #16: MCP-2515 not supported for ESP32

tamusjroyce, 2019-03-18:
Great libraries. But it doesn't support MCP-2515 for ESP32. Not complaining one bit.
But I was a little confused at first. And wanted to leave a note for others.

MCP-2515 would be an awesome feature. But this library is already good enough as-is. Thank you for the awesome work!

Petros144, 2019-04-22:
works straight away with the SJA1000 build in the ESP32.

tamusjroyce, 2019-04-23:
Really? With the mcp-2515? I went ahead and bought the Waveshare SN65HVD230. And this library works great with it. But I would love to utilize two can bus's (translate between two bus speeds).

master (at least was) setup so that if you are using an arduino, it would use the MC_2515. And for the ESP32, it would use the transceiver-only SN65HVD230.

That is very good to know @Petros144! Thank you!

yahyatawil, 2020-02-10:
you can edit the library code to make MCP2515 files works with ESP32. Just notice `#ifdef ARDUINO_ARCH_ESP32` in the beginning of each source file which will select automatically if CAN instance is from MCP2515 or SJ1000. Some minior changes maybe needed in MCP2515 like the need to specify SPI number (hspi or hspi).

HJuliamPR, 2020-05-06:
Yeah, I tried to modify to make work the ESP32 with the MCP2515, but there are some functions like  SPI.usingInterrupt(digitalPinToInterrupt(_intPin)) and SPI.notUsingInterrupt(digitalPinToInterrupt(_intPin)) that are only avaiable in AVR microships.
"https://github.com/pierremolinaro/acan2517/issues/1"
In the link you could see how they solved a similar problem, however, I wasn´t able to use it in arduino-CAN Library.
if someone can solved it, please let us know how do it.
Good day, by the way.

lordmundi, 2022-08-06:
I think I ran into this same problem. I have an ESP32 dev board (which I don't think has any CAN interface on it) and I have an MCP2515 hooked up on the VSPI bus. When I try to use this library, the #ifdef is trying to take a different path I believe. I tried the following, but it just let to errors about undefined references:

`
#undef ARDUINO_ARCH_ESP32
#include <CAN.h>
#define ARDUINO_ARCH_ESP32
`