sandeepmistry/arduino-CAN #21: Same packet received over and over (ESP32 and 500Kbps vehicle bus)

tmkdev, 2019-09-02:
Tried the basic receive example after trying the OBD keystats example with no success.

Seems to be getting valid data from the bus - but keeps printing the same packet over and over. Same ID, same packet data. Like the buffer is not being cleared?

tmkdev, 2019-09-02:
I have been able to duplicate this - seems to happen when the bus load is high - cangen can0 -g 3 from a raspberry pi to the ESP32 seems to cause the issue. 5 ms packet gap is stable.

Schmackhaft, 2019-11-08:
I am having the same problem. Have you been able to find a soluction?

yogeshwaran774, 2019-12-09:
> 
> 
> I have been able to duplicate this - seems to happen when the bus load is high - cangen can0 -g 3 from a raspberry pi to the ESP32 seems to cause the issue. 5 ms packet gap is stable.

i am also having the same problem. basically my device device was running properly around 8 hrs after that i am keep on getting same msg in can receiver side and it never stops. Did you find the solution?

tmkdev, 2019-12-09:
Never really fixed it - kept my project from being finished. I'll dig into this again maybe over the December break.

beikeland, 2020-01-11:
Same here. Using a SN65HVD230 module, built in 120ohm termination, Rs=10k sets the chip to slope control. Might have a go at replacing the 10k with a short to ground to put the module in high speed mode. EMI is not a concern for just listening anyway.

panakos, 2021-05-16:
Just put a 10+ ms delay in your loop

jbraceg, 2021-10-16:
Did you ever figure this out? I'm having exactly the same problem. Did shorting pin 8 to ground for high speed mode help?

panakos, 2021-10-16:
> Did you ever figure this out? I'm having exactly the same problem. Did shorting pin 8 to ground for high speed mode help?

I used a 10ms delay in the loop to prevent the loop from reading the same package twice.

It's better to use micros rather than delay, but delay will also work

jbraceg, 2021-10-17:
Tried adding a 10ms delay, it didn't help.

timurrrr, 2021-10-17:
I strongly suggest not trying to "fix" bugs by adding `delay`s.
It's not going to fix anything, it will only make the issue less likely to happen.
The issue will still happen, possibly when you least expect it.
Since we're talking about a library with automotive applications, I don't think that's good enough.

panakos, 2021-10-17:
> I strongly suggest not trying to "fix" bugs by adding `delay`s. It's not going to fix anything, it will only make the issue less likely to happen. The issue will still happen, possibly when you [...]

The delay is because the esp32 scan cycle is much faster than the comms update rate , so it reads the same package in multiple scans. Delay is an easy way, best way as I said earlier is to use micros so to chose how often to read. 

On the other hand this are diy project mostly that we only read not write to the canbus , so I can't see any harm

fightforlife, 2022-06-04:
Same issue here, sadly.