sandeepmistry/arduino-CAN #25: No RTR receiption

aviatorhh, 2019-09-25:
Unfortunately RTR messages are not handled correctly. If you look at

```
void MCP2515Class::handleInterrupt()
{
  if (readRegister(REG_CANINTF) == 0) {
    return;
  }

  while (parsePacket()) {
    _onReceive(available());
  }
}
```

you can see that `_onReceive` will only get called if `parsePacket` returns non-zero.

If the function `parsePacket` detects the RTR bit set it sets the _rxDlc (data packet length) to zero which in terms of RTR is correct. Unfortunately it returns that zero value.

`return _rxDlc;`

The solution is simple. Modify the return statement in `parsePacket` to:

`return _rxRtr ? true : _rxDlc;`

BiggRanger, 2020-01-01:
I'll give this a try in the next few days. I was expecting some RTR packets, but never saw them (using the callback onReceive).

sandeepmistry, 2022-11-19:
PR https://github.com/sandeepmistry/arduino-CAN/pull/103 might help with this.