sandeepmistry/arduino-CAN #30: Missing 83.333 kbps

BiggRanger, 2020-01-01:
The mode for 83.333 kbps is missing. I added the following line for the 16MHz clock:
{ (long)16E6,   (long)83E3, { 0x03, 0xfe, 0x87 } },
in MCP2515.cpp

These settings work and were tested in my 2008 Dodge Durango poking around CAN-IHS.

I've ordered a few more modules, one with an 8MHz clock to verify the settings for 83.333 kbps, I should have them sometime in the beginning of January 2020.

BiggRanger, 2020-01-12:
83.333K is tested on modules with 8MHz and 16MHz crystals. Here is the updated code snippet for MCP2515.cpp

On all the 16MHz CNF[3] parameters I knocked the MSB off the value since nothing is using the SOF pin on the MCP2515.

```

    uint8_t cnf[3];
  } CNF_MAPPER[] = {
    {  (long)8E6, (long)1000E3, { 0x00, 0x80, 0x00 } },
    {  (long)8E6,  (long)500E3, { 0x00, 0x90, 0x02 } },
    {  (long)8E6,  (long)250E3, { 0x00, 0xb1, 0x05 } },
    {  (long)8E6,  (long)200E3, { 0x00, 0xb4, 0x06 } },
    {  (long)8E6,  (long)125E3, { 0x01, 0xb1, 0x05 } },
    {  (long)8E6,  (long)100E3, { 0x01, 0xb4, 0x06 } },
    {  (long)8E6,   (long)83E3, { 0x01, 0xbe, 0x07 } },
    {  (long)8E6,   (long)80E3, { 0x01, 0xbf, 0x07 } },
    {  (long)8E6,   (long)50E3, { 0x03, 0xb4, 0x06 } },
    {  (long)8E6,   (long)40E3, { 0x03, 0xbf, 0x07 } },
    {  (long)8E6,   (long)20E3, { 0x07, 0xbf, 0x07 } },
    {  (long)8E6,   (long)10E3, { 0x0f, 0xbf, 0x07 } },
    {  (long)8E6,    (long)5E3, { 0x1f, 0xbf, 0x07 } },

    { (long)16E6, (long)1000E3, { 0x00, 0xd0, 0x02 } },
    { (long)16E6,  (long)500E3, { 0x00, 0xf0, 0x06 } },
    { (long)16E6,  (long)250E3, { 0x41, 0xf1, 0x05 } },
    { (long)16E6,  (long)200E3, { 0x01, 0xfa, 0x07 } },
    { (long)16E6,  (long)125E3, { 0x03, 0xf0, 0x06 } },
    { (long)16E6,  (long)100E3, { 0x03, 0xfa, 0x07 } },
    { (long)16E6,   (long)83E3, { 0x03, 0xfe, 0x07 } },
    { (long)16E6,   (long)80E3, { 0x03, 0xff, 0x07 } },
    { (long)16E6,   (long)50E3, { 0x07, 0xfa, 0x07 } },
    { (long)16E6,   (long)40E3, { 0x07, 0xff, 0x07 } },
    { (long)16E6,   (long)20E3, { 0x0f, 0xff, 0x07 } },
    { (long)16E6,   (long)10E3, { 0x1f, 0xff, 0x07 } },
    { (long)16E6,    (long)5E3, { 0x3f, 0xff, 0x07 } },
  };
```

commcad, 2020-02-17:
Nice one, thanks for posting the solution @BiggRanger

demigh0d, 2020-03-09:
Can anyone provide the register settings for ESP32SJA1000.cpp?

Been trying to figure it out myself but have had no luck.

pamm01, 2021-05-06:
Hello, how to add 47.619 kbps ?