sandeepmistry/arduino-CAN #35: Processing suddenly stops after 1 second at 500kbps and 1000kbps

dARoB2108, 2020-02-21:
Hi!

I have wired an sn65hvd230 to the ESP32s standard CAN RX and TX pins and I'm using the standard library with standard receive example where I change only the baudrate for testing.

As a CAN sending device, I have a notebook with Busmaster and a logfile which I'm replaying and sending through a Vector CanCaseXL.

When setting up the standard example as well as the sender to 100kbps, everything works fine, no glitches, no isses.
The story changes when changing the baudrate to 500 or 1000 respectively.
It takes approx 1s (+10-100ms) until the communication stops.
I monitor the communication through the serial monitor in ArduinoIDE.

Since my level is beginner, I first started looking if someone had similar issues but found nothing, hence why I'm writing this and asking for help on how to proceed further and how to troubleshoot what might be the issue? I have a small oscilloscope to monitor stuff if this is needed.

Thanks!

Best regards,
RoB

dARoB2108, 2020-02-24:
Would be nice if someone can try out the above with the sn65hvd230 CAN transceiver as now I'm pretty sure that (at least in my case) the transceiver is the whole problem.

I received a TJA1050 today and it works like it is supposed to. It was literaly exchange one against the other, start the CAN stream and everything worked nicely.

BR
RoB

Petros144, 2020-03-16:
Hi Rob! I work with the same boards on Proto Boards and also usee the same chip in my product. No problems so far! 

you can Probe Volatages maybe this is the prblem?

I also want to mention that the Board also comes with 120 OHM resistor on board (I desolder theese) because my environment is already terminated. Im not shure about your test setup by try to remove the resistor and give it a try! Baud over 500kbs needs good connections -> TP cable .

dARoB2108, 2020-03-16:
Hi!

Thanks for the heads up but due to complete waste of time, I have gone on with the other chip ehich works flawlessly!

Regarding the resistor, I have tried the board with and without it so this wasn‘t the issue. Also, I have monitored the voltages which were fine so this could not be the issue as well. As mentioned, exchanging the sn65 board to the other was plug and play and without a change in the code everything was running smooth and without issues!

Regarding the wiring for the bus speed above 500kbps, I cannot agree.
I‘m working with the canbus for the past 10 years and for all my sniffing and even for my complete car rewire, i never used a special cable, also, i do not twist it as well as it is not necessary for lengths under 20-30m.
If it is terminated at both ends, everything will be fine. My test setup is on my working desk and a max of 1m in length! Bare in mind that the oem car manufacturers also don’t use special cables, most of them are not even twisted.

Thanks!

Best regards
RoB

Petros144, 2020-03-17:
Nice To hear, I made some investigations and found out that some Chineese Chips are Fake so maybe you got your hand on a bad Copy of silicon.

wow okay, this is new to me. Most german cars I saw used Twished pair cables for Can bus signals, maybe for signal integrity who knows.  

Maybe you can tell which Chip you used for other People with the same issue?

timurrrr, 2020-07-31:
When you controller gets into that state, what are the values of the EFLG, TEC, REC registers?