sandeepmistry/arduino-CAN #36: Multiple CAN instances

reubenfinch, 2020-03-02:
There does not seem to be a way to create more than a single instance of the CAN object.

Petros144, 2020-03-16:
why? you only have one on-board Can interface? SJA1000

j4m3s, 2020-07-30:
@Petros144 it's a reasonable scenario, where you have multiple MCP2515s or an MCP2515 over SPI attached to an ESP32.  I'm looking to do the same in order to monitor two CAN buses.  Is it possible?

aaknitt, 2021-04-12:
It looks like it can be done by going down a layer and creating instances of the MCP2515 class directly.  Make sure to call setPins() before begin() for the instance that's not using the default pins for chip select and interrupt.

`void setup() {
  MCP2515Class CAN1;
  MCP2515Class CAN2;
  CAN1.begin(5000);
  CAN2.setPins(3,4);  //set pins to use for SPI Chip Select and CAN interupt for our second CAN controller
  CAN2.begin(5000);
}`

TheCooolNerd, 2022-07-24:
> It looks like it can be done by going down a layer and creating instances of the MCP2515 class directly. Make sure to call setPins() before begin() for the instance that's not using the default pins for chip select and interrupt.
> 
> ```
> void setup() {
>   MCP2515Class CAN1;
>   MCP2515Class CAN2;
>   CAN1.begin(500000);
>   CAN2.setPins(3,4);  //set pins to use for SPI Chip Select and CAN interupt for our second CAN controller
>   CAN2.begin(500000);
> }
> ```
Hi. This works if using one spi lane by defining the cs and int pins. what if i want to assign the second can instance to a different lane with seperate MOSI, MISO and SCK lines?