sandeepmistry/arduino-CAN #44: Even example code doesn't always work

ppeponis, 2020-07-24:
I have made the setup you show us. 
I can use the examples, like read the rpm and read the vin code etc. Although the examples doesn't always work. For example the rpm doesn't always show in the console. My connections am pretty sure they are good. Do you have any suggestion?  
Also when i try to read different PID i cant make it work either. I try simple PID to keep the code same as the RPM example and if it works to move forward to something more complicated.  I try to read MAF sensor PID since it is very similar to RPM PID. 
So i changed the code from the RPM example          CAN.read() != 0x0c       // correct PID
  RPM PID   to this   CAN.read() != 0x10 since 0c is the rpm hex and 10 is the maf value and also change the calculation algorithm to match . But it doesn't work i get no data at all.

timurrrr, 2020-07-29:
I suggest building a histogram of all the PIDs you're receiving, printing and resetting it every N seconds, and compare to the list of PIDs that you have for your vehicle.

ppeponis, 2020-07-29:
The PIDs i try to use are basic. and definitely are supported by my ECU. RPM example works some times. I have to reset the arduino and car several times in order to see the RPM in the console. My connections are good. Because once it works it won't stop until i reset the car or arduino, then stops working.

timurrrr, 2020-07-31:
Ah, perhaps this should be filed against https://github.com/sandeepmistry/arduino-OBD2 now that it's a separate library?

ppeponis, 2020-08-03:
I used the library you suggested and those examples like RPM always work. Thanks!!

closed 2020-08-03 by ppeponis