sandeepmistry/arduino-CAN #52: Cannot get CAN library and SPI display (ILI9341) working at the same time

RBVGH, 2020-08-26:
I am working on an Arduino MKR Zero based project that combines an MKR CAN Shield as well as a touch screen (2.4” 240x320 using the Adafruit ILI9341 library). 

When connected alone, the CAN Shield works fine using the Sandeep Mistry's arduino-CAN library and the CANSender code. When connected alone, the display assembly also works fine. 

But I do not get both parts working at the same time. When running CANSender, it gives a “Starting CAN failed!” error message in SerialReader.

I checked whether there was a conflict on resources. The CAN board uses SPI with Chip Select (CS) on port 3, the display and touch panel use SPI with CS on pin A3 and A4 respectively. So, there should not be any conflict of resources there. The CAN board also uses interrupt on Pin 7, which is not used by the display or touch panel. 

Interestingly, when I power up with just the CAN shield first, wait for successful initialisation, and THEN connect the display module, the CAN bus traffic continues to work fine (again using just CANSender, no code for the display at all). But when I power up with both connected, everything else equal, I get the above error. So it seems only the initialisation phase that has problems, not the later communication phase. 

Am not really a programmer… Anyone any clue or suggestion how to get this working? 

Thanks!!

timurrrr, 2020-08-27:
A good starting point would be to figure out what exactly fails.
Try opening the source code of the arduino-CAN library installed into your Arduino IDE, follow the code that you call before you get the failure, and insert something line `Serial.println("Failed doing XYZ");` at each block of code like `if (not ok) { return ...; }`. This way you should be able to find out which condition fails.
This is how I found the bug here: https://github.com/sandeepmistry/arduino-CAN/pull/45

RBVGH, 2020-08-28:
Thanks Timurrr for the reply. I should indeed try that, but it's rather complicated as the CAN library in turn calls other libraries (such as those for the SPI bus) and the problem may be there.

I also have another lead: the SCK signal looks rather degraded when the display and touch chips are connected to the SPI bus (voltage drop, reduced slew rate). Just connecting these to the SCK bus even when MISO and MOSI are disconnected results in an initialisation failure of the CAN shield. I may need to look further (regenerate clock by an additional chip, or reduce clock rate).

Petros144, 2020-09-06:
What kind of CAN tranciver /controler are you useing? I also work with the same TFT Driver in my project and have zero problems.

I use the onboard ESP32 Can Controller and a SN65HVD230 tranciver... this may be a better solution for your purpose?

timurrrr, 2020-09-06:
> Just connecting these to the SCK bus even when MISO and MOSI are
> disconnected results in an initialisation failure of the CAN shield.

Yeah, sounds like you need to figure out a way to better isolate them electrically.