sandeepmistry/arduino-CAN #53: CAN bus fault detection

aklein9999, 2020-09-06:
Hello :-)

Would it be possible to implement some standard signals concerning CAN bus fault detection?
E.g. Transmit Error Counter (TEC), Receive Error Counter (REC) and  the typical CAN bus Error Flags?
For me your CAN bus driver in connection with the ESP32 works very well, thank you very much for providing it to all of us !!!

Alex

timurrrr, 2020-09-06:
Should be simple to implement.
You can follow https://github.com/timurrrr/arduino-CAN/blob/7d0fd5d8a4bdfa0e91af6964f7c83de2de91b0f5/src/MCP2515.cpp#L555 from my fork of this library for pointers on how to read those things via SPI.

aklein9999, 2020-09-06:
Dear timurrrr,
Thank you very much for your quick reply.
I am using the internal SJA1000 of the ESP32.
I think the registers here are different, aren't they?

timurrrr, 2020-09-06:
Oh sorry I didn't notice the last paragraph in your original message 🤦‍♂️

mariliahoshino, 2020-10-13:
I'd like make a similar project for count Receive Error Counter (REC) , error frames, CRC
it's possible analyser cable with bad connection cause disturbs in CAN network 
Tks

mariliahoshino, 2020-10-17:
I'm using internal SJA1000 too

aklein9999, 2020-10-18:
I have been buys for a while with another project but I would like to come back to the old issue. If somebody could provide a code or a simple explanation how to access the error flags? It would be really helpful. Thank you in advance :-)

mariliahoshino, 2020-10-20:
there's this datasheet
https://www.nxp.com/docs/en/data-sheet/SJA1000.pdf

I don't have idea yet, how calculate CRC and REC, error frame

mariliahoshino, 2020-10-21:
https://github.com/mariliahoshino/CAN_Interface_board
 
I'm making a project CAN Analyser 
I need check error frames CRC

aklein9999, 2020-11-22:
For some unknown reasons the CAN driver hung up in my application. Exactly at the moment when I called CAN.beginPacket(CAN_Tx_id, CAN_Tx_len) to send a message.
Sometimes it happens, sometimes not. It might have a correlation with the WiFi mode which I also use in that application.

Anyways, if someone has a similar problem, it seems that I  have found a solution for this.
Just run the whole application on the ESP32 core 1, and the CAN sending part on core 0. Now the application does not crash anymore. I am not sure why this works, maybe someone has an explanation?

I have another question to timurrrr. 
As we can currently not read the CRC, REC and error frame (unless mariliahoshino found a way) with the sandeepmistry driver, I feel a bit unsafe of using it in applications with more than two or three CAN nodes.
So I looked a bit deeper into the ESP32 CAN errata sheet:
https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf
---> chapter 3.13 ff

Are these issues implemented in the arduino driver? Or are we fully facing them when we use it?
I could program around these issues as described in the errata sheet. But then it would be necesary, e.g., to:
- set LISTEN_ONLY_MODE
- check STATUS_NODE_BUS_OFF
- detection of bus-off recover
- check STATUS_TRANSMIT_BUFFER 
- and so on....

In the ESP32 documentation:
https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf
I only found registers for the TWAI interface. Are these the registers I should use?

One detailed example how to access the CAN bus registers in the ESP32 would be great! :-)

Thank you!

P.S.:
In general I believe it actually is really a great thing to use the CAN controller within the ESP32 (of course only if it can work reliably). Together with the WiFi interface, EPS NOW and the touch button capability, in combination with its price - this chip is most suitable for IOT applications where multiple interfaces are necessary. And sometimes a wired interface has its advantages and is needed, and for control systems, nothing beats the CAN bus.

amaiaMDNURRA, 2024-10-23:
Hello!
Did anyone achive a CAN communication error counter? I need it on my proyect with ESP32 and internal SJA1000,
Thank you!