sandeepmistry/arduino-CAN #54: ESP32 with internally used SJA1000 hangs completely up when CAN wires are shorted

aklein9999, 2020-09-07:
Dear all,

In my application, the ESP32 completely gets stuck when I short the CAN wires (CAN-J and CAN-L connected).

Petros144, 2020-09-15:
im not shure if it is a good idea to short the wires anyways... 
if your project is made for this kind of Errors why not use some isolation IC with build in CAN tranciver?

does this error occure when you want to send a message or while reading the bus?

aklein9999, 2020-09-15:
Dear Petros144,

Shorting the wires is just a fault simulation. Normally the microcontroller should return to normal operation and not completely hang up (the watchdog timer cannot reset it directly from the Arduino programming environment). However, I have connected an ESP32 output pin to the external ESP32 reset pin, now I can reset it reliably.

I think the main issue is that I cannot catch any generated CAN error code (once the CAN error counter counts up) as the error codes of the internal SJA1000 CAN controller are not supported at all yet. I have mentioned this in the previous issue.

I haven't tried yet, but I believe that an isolated CAN node will make no difference. Once the bus is shorted the error counter will be at its limit at no time and the ESP32 will hang.

Thank you for your input,

Alex

Petros144, 2020-09-15:
maybe this helps?

https://github.com/espressif/esp-idf/issues/4276

Petros144, 2020-09-15:
also this is maybe interresting:

http://www.barth-dev.de/can-driver-esp32/