sandeepmistry/arduino-CAN #56: Sending and receiving

janderkan, 2020-11-01:
Sending and receiving works fine.
I have several ESP32 on the bus and they are programmed to send a packet when receiving another.
So 1 ESP sends packet number 2 and all other ESP sends packet number 3 when receiving packet number 2.
This makes f.ex 10 ESP send the same packet at the same time.
If I repeat this , then 1 ESP eventually dies, and after a while only the sender and 1 other is working.
The ESP are dead and need power on/off to live again.
I am thinking receive buffers, anyone experience the same ?

Teddyz, 2020-11-01:
CAN-bus is good at handlig many senders at the same time. But with one exception. If two or more senders try to send a frame wirh the same ID at the same time both will fail, causing an 'error frame'. 

Best practice is to not come into this situation in the first place, so use one or more IDs for each sender.

I don't know what went wrong in your design, but it might be that you need to clear the error-frame-error for each node. 

In a car, as an example, a node that cause many error-frames must go offline to avoid ruining the bus for all other nodes. This can be what happened to you.

I have no specific knowledge of this library.

Petros144, 2020-11-04:
and also be notified that lower ID´s are higher prioritized than the higher ones.

like ID 10 will block ID11,12.... and so on.

janderkan, 2020-11-04:
Thank you both, this solved my problem.

closed 2020-11-04 by janderkan