sandeepmistry/arduino-CAN #58: Change RX/TX pins

craig-timms, 2020-12-28:
Is there a way to configure different RX/TX pins?

Petros144, 2020-12-30:
maybe have a look at ESP32SJA1000.h

line 11,12 

#define DEFAULT_CAN_RX_PIN GPIO_NUM_4
#define DEFAULT_CAN_TX_PIN GPIO_NUM_5

try some other pins... it should work....

craig-timms, 2021-01-01:
I've tried this on RX=32/TX=33 and RX=25/TX=26.

I've also tried changing the pins in the ..esp32\hardware\esp32\1.0.4\cores\esp32\HardwareSerial.cpp (https://hackaday.com/2017/08/17/secret-serial-port-for-arduinoesp32/)

With no success. Any other ideas?

I have the same hardware that is using this library successfully with another ESP32 and the standard pins 16/17.

Petros144, 2021-01-01:
Do you want to interface via serial connection? im not shure what you want. 

what is the problem using default pins? for CAN bus?...

craig-timms, 2021-01-02:
So I'm using the lib and the standard RX=16/TX=17 lines connected to the SN65HVD232 transceiver without issue. This is with the ESP-32 board below.

[image]

I also want to use the lib with this board. This board does not have the standard serial pins available though. Therefore I need to interface with the SN65HVD232 using other RX/TX lines. Ideally I'd like to use X=32/TX=33 or RX=25/TX=26. I've used 25/26 before as serial lines with another library by using this library https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp

[image]

Petros144, 2021-01-03:
im still not shure what you want to do with the Serial Interface with a CAN tranciver?!?  
This Lib is for Can Bus and are default to use with GPIO 4 & 5.

Pins 16/17 are normaly used for Serial1

Please tell us what exactly is your plan so we can move on. 

If you have problems with the Serial infertace look for SoftSerial emulation.... this works on any pins.

craig-timms, 2021-01-04:
Should have looked closer at the readme, sorry. This is the answer to my question: CAN.setPins(rx, tx). Not working yet but will update.

closed 2021-01-04 by craig-timms