sandeepmistry/arduino-CAN #63: Can’t get this library working with esp32

alexvnesta, 2021-02-15:
I have this up and running with my arduino uno and mcp2515. 

I tried the to hvd230 and hvd232 transceivers with my esp32 but I can’t get either of them to work.

does anyone know where I can purchase a can transceiver that works with this library for the esp32? Does The the transceiver chip number matter? For example, Amazon has mcp2551 and tja1051. Would any changes in the code be necessary to use these transceivers instead of the recommended hvd230?

thanks,
Alex

Petros144, 2021-02-15:
Hi Alex,

I just bought the cheapest SN65HVD230 Breakout boards i can find on Ebay and they do their job. do you have the Connections right?
What kind of errors do you get?

alexvnesta, 2021-02-15:
https://imgur.com/gallery/QIhNPT6

Uploaded info to imgur. To summarize, I show the library working fine with an Arduino UNO and MCP2515 connected via SPI, then I show two examples wired up to the esp32 - VP232 chip and VP230. Both show no messages being received on serial monitor.

Voltages read by multimeter on vp230 and 232 are the same - 3.3v on 3.3v, 0 on GND, CTX  3.3v, CRX 1.8v, CANH 1.3V, CANL 1.8v.

I can try hooking up a logic analyzer to the CTX and CRX next.

Petros144, 2021-02-17:
@alexvnesta I think you got some broken IC´s  Best Chip to use is the VP323, it has no Rs Pin and is the most basic one.
MCP also has some Trancivers but they are all 5V VCC/Logic. 

Just get a new set and test them out in a controlled environment (for example Two ESP32 sending/reciving data).

Did you test out if something comes out of the can lines and ESP GPIO´s if you send a Packet out?

alexvnesta, 2021-02-18:
I hooked up my MCP2515 to my Arduino and set up the send hello world over can. The VP230 is on the receiving end with the ESP32. This is working fine!

This is very strange, why does sending data from the MCP2515 to the VP230 work, but receiving data from my radio to the VP230 does not work, even thought I am using the same 100k baud rate?

Even stranger, I am able to receive data from the radio using the MCP2515 on ardunio, but not with the VP230 on ESP32.

CAN baud rate is set to 100kbaud for everything - test packets and real packets. Are the other parameters that might need to be changed?

What is different between the two chips that allows the MCP2515 to receive real data, but the VP230 cannot?

Petros144, 2021-02-19:
I dont think that it has to do with the Hardware itself. Did you desolder the Terminating resistor on the VP320 Board? maybe you get some erros with it.

and how did you catch the pakets in your programm?
I would suggest to directly read out the raw TX buffer to have plain 8 bytes. The examples are a bit wierd.

Alex-Nesta, 2021-02-24:
This issue can now be closed.

1) The library works fine with ESP32.
2) All can receivers work with normal can protocol.
3) It is important to note that fault tolerant can is used in some automobiles, and this requires a fault tolerant transceiver.
4) I ended up getting this library to work with both a TJA1054T and A TJA1055 - both fault tolerant transceivers. No special coding was required. Just had to get the right chips.

joesoh, 2021-02-26:
> This issue can now be closed.
>
> 1. The library works fine with ESP32.
> 2. All can receivers work with normal can protocol.
> 3. It is important to note that fault tolerant can is used in some [...]

Hi Alex, thanks for sharing the updates. I have the exact same issue you encountered - the same code works on Uno + MCP2515 but message not receiving on ESP32 + SN65HVD230 with right pins configured. 

Do you mind to share TJA1054T and TJA1055 link you used, I would like to give it a try too.

alexvnesta, 2021-02-26:
Hi Josoh,
Before you order the chips, try the following

1. send the hello world test program from the arduino to the esp32.

If that does not work, there is either an issue with your wiring or the hvd230 - some hvd230s can be bad.

if that does work, check what can transceiver the device you are trying to communicate with is using. In my case, the device was using a tja1054t - a fault tolerant can transceiver.

Then find a chip that is compatible with the can transceiver and buy it. I ended up buying a tja1054t and a surface mount to pin header. I read the data sheet to find the pouch between the surface Mount pins to get the correct adapter. Then I followed the application diagram to find the correct pins for wiring. 

Hope that helps.
Alex

closed 2021-02-26 by alexvnesta

Petros144, 2021-02-26:
Thank you for the information! @alexvnesta.

For some more information, why the Bus goes Offline see Page 18 & 19 in this thesis Document, it explains is verry well.

https://cta.physik.uzh.ch/public/theses/files/2014-WidmerTimothy-Bachelor.pdf