sandeepmistry/arduino-CAN #66: High speed sender

atkforever, 2021-03-11:
I'm trying to send packets every 10ms but I get this on the serial monitor, few packets are sent in a row, then it stops few ms

12:38:27.527 -> Sending packet ... done
12:38:27.527 -> Sending packet ... done
12:38:27.527 -> Sending packet ... done
12:38:27.527 -> Sending packet ... done
12:38:27.540 -> Sending packet ... done
12:38:27.540 -> Sending packet ... done
12:38:27.574 -> Sending packet ... done
12:38:27.574 -> Sending packet ... done
12:38:27.574 -> Sending packet ... done

Teddyz, 2021-03-11:
Not much info here. Looks as you actually tries to send more often than every 10 ms. You need a receiving node to successfully send any CAN-frame.

Petros144, 2021-03-25:
Do you have any traffic on the Bus from other nodes? also Lower ID´s are prioritized and your node is on a hold state.

you also you need to terminate the bus properly to work as expected.