sandeepmistry/arduino-CAN #70: CAN will not work on M5Core2?

joshuavaneck, 2021-04-12:
Hii!

I used changepins (32/33)

M5core2 does "see" the canbus module, but does not communicate with the car..

If i try the same on m5stack grey (pins 16/17)
It works perfect!

Anybody ideas??

Thnx!!

Petros144, 2021-04-14:
You need to use proprietary Can Pins on the Board.  IO 4 & 5 must be M5 Pins 17/18? I dont work with the m5. I dont think you can use other pins.

joshuavaneck, 2021-04-14:
No ubr incorrect, and since u dont work with m5 why the hell u respond?

READ:  the module does communicate with m5core2 but not with car

mister0wl, 2021-04-14:
Pls look at issue #62, maybe your chip on board is v2 or higher revision so the problem is with speed setting. The easiest method to test is if you double the speed, e.g. try 1000E3 instead of 500E3. If it works then you have the same "feature" (Espressif changed the spec so the interrupt register bit.4 changed its meaning). Fix is on the way...

joshuavaneck, 2021-04-15:
I tried ur suggestion and it works!

Setting to 1000E3 did the trick! 
For now i can use it like this,
Cant wait for the fix!

Tobbera, 2023-01-19:
> Pls look at issue #62, maybe your chip on board is v2 or higher revision so the problem is with speed setting. The easiest method to test is if you double the speed, e.g. try 1000E3 instead of [...]

I can confirm that this also works for me. The solution is to either:
1. Use the modified library at [](https://github.com/mister0wl/arduino-CAN)[https://github.com/mister0wl/arduino-CAN](url)
OR
2. Define `CAN.begin(1000E3)` when you actually want 500 kbs baud.

Are this going to be added into the actual main library? That would have spared me 8 hours of troubleshooting.

italocjs, 2023-10-18:
Hello, did anyone managed to make this library work with chip ESP32-D0WD-V3 (revision v3.1) ?
It works very well using the TWAI from ESPIDF,  but this library doent start, it gets stuck and doenst seem to be able to start the can bus.  Im using pins 18/19 and i have modified the library to use these pins, to no success.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
OBD2 Supported PIDs
Attempting to connect to OBD2 CAN bus ... (stuck here forever)
```