sandeepmistry/arduino-CAN #72: How to verify SN65HVD230 is somehow functional

maxdd, 2021-04-24:
Hi,
recently i bought an SN65HVD230 from amazon but looking around it seems there are plenty of non-working/fake-devices.
Currently with the standard receiver example i can't read any data on my motorbike project (ELM327 works fine).
The SN65HVD230 board has a 10K Ohm between GND and VCC and a 120 Ohm between CANH and CANL.
I have also read about ppl puttin 4k7 Ohm pull-up resistor on CTX and RTX pin.
I'm wondering before returing/soldering if i have a way with this library to e.g. read some registers which allows me to understand whether a connection with SN65HVD230 is effectively in place or not.
Is an oscilloscope the only solution?

Regards,

Petros144, 2021-04-30:
Hi Max,

There is a issue of the Baud rates with some newer ESP32´s maybe try the next higher baud rate and see if you get imput signals on the Can lines.

ami3go, 2021-04-30:
Hi. Also the easiest way is to have 2  mcp2515 (spi CAN chip) and two arduino boards (transmitter and receives).. Connect your board to CAN network and you will see

zalexzperez, 2021-06-12:
I won't be able to help you with your doubt. I'm also interested in this topic, though.

Just for your information, there's an Aliexpress store called Lovesell2013 Store that sells waveshare module clones that work just fine. They look like this:
[image]

As a matter of fact, I built my own PCB for my project, bought some SN65HVD230 chips and they didn't work at all, they were fakes. I had to unsolder the chips from the modules I already had.

maxdd, 2021-06-12:
Hi,
Thank you for the suggestion.
I started to build my PCB as well but I ended up desoldering the apparently broken one with a new one I bought from a local trusted reseller. Where did you buy yours? I was wondering whether jlcpcb or similars have the real one or not

zalexzperez, 2021-06-13:
The CANBUS modules I bought on several ocassions were from Aliexpress, [here's the link](https://es.aliexpress.com/item/4000376804990.html?spm=a2g0s.9042311.0.0.79d563c0yrKY0X) .

When I was considering using jlcpcb's SMT service I searched for the SN65HVD230 chip but they were sold out except for a SN65HVD230Q-DR device at LCSC electronics. 

However, I'm unsure whether those 'Q' chips would work with Arduinos or not as the datasheet states: "The **SN65HVD230Q**, SN65HVD231Q, and SN65HVD232Q controller area network (CAN) transceivers are designed for use with the Texas Instruments TMS320Lx240x 3.3-V DSPs with CAN controllers, or with equivalent devices."

"The **SN65HVD23x** transceivers are designed for use with the Texas Instruments 3.3V µPs, MCUs and DSPs with CAN controllers, or with equivalent protocol controller devices"

The safest and cheapest option for me was to create two soldering blobs at each side of the chip and use two soldering irons to remove it from the PCB.

sysavm, 2023-01-18:
Had the same experience here. Bought 2 from amazon cheap and tried sandeep's and some ESP32CAN library and no luck to get the module working.

Is the MCP2551 chip cross compatible with this module? Thinking of replacing the SN65HVD23x  with it.

zalexzperez, 2023-01-18:
> Had the same experience here. Bought 2 from amazon cheap and tried sandeep's and some ESP32CAN library and no luck to get the module working.
> 
> Is the MCP2551 chip cross compatible with this module? Thinking of replacing the SN65HVD23x with it.

These look genuine: https://www.aliexpress.com/item/1005002843997801.html

The store I was buying them from no longer sells it. It sells it in a different board design, I'm unsure they're genuine or not... [they're here](https://www.aliexpress.com/item/4000376830862.html).

Your other choice is to buy them from LCSC.