sandeepmistry/arduino-CAN #75: Not Working with ESP Wroom32 E

SomenKumarDas, 2021-06-05:
Hi I've using your code for my wroom32D module which is working very fine. But I got a new esp wroom32E module but the can is not working there at all. Can you help me in that...

jcernato, 2021-06-12:
Allthough having some minor issues not receiving all packets correctly, generally my wroom32E with SN65HVD232 works fine.
Maybe you'd like to share some information about your setup?
I'm working with platformio(vscode) on ubuntu, arduino framework.
The transceiver chip is connected to GPIO pins 4 and 5 as the lib suggests. With the SN...23**2** I had to connect pin 8 (RS) to ground in order to work.

SomenKumarDas, 2021-06-12:
I'm using MCP2561-E/SN as the can transceiver, Now what I got here with testing... 

I've seeing that if I set 500k Baud for wroom32D its CAN working in 500, But in Wroom32E its working in 250k, Any idea why its behaving so?...

Clock frq is same as 240MHz,

altoriot, 2021-06-16:
@SomenKumarDas I may be commenting out of step but there is a problem with the 32E chips that the register bit regarding can is set in a half state. in my code, can at 500k needs to be started at 1M , so if you want 1M can speed, start at 2000E3 and you will get can at 1Meg. i belive theres another gentleman goes by owl that fixed the driver so that it works as per normal speed, I implemented his fork and it sorta works. but I still have endless hassles with the driver where if ESP reboots and there is still data in the buffer, the "treadmil" as I call it goes out of step and I get garbage ID's and data.

SomenKumarDas, 2021-06-17:
@altoriot I don't know about 2000E3 there is nothing like it.. Can you be more specific about it... and help me on that... I need to implement the 1Mbps baud

altoriot, 2021-06-17:
Because of the weirdness of the register issue, if you call CAN.begin(2000E3), the can baudrate will start at half the baudrate specified, in this case, half of 2meg, which is what you are looking for. its a lowsy workaround, but its how I've used it with success

SomenKumarDas, 2021-06-17:
In the begin function there no 2000E3 to call... begin functions return fails..

altoriot, 2021-06-17:
Try this. pull https://github.com/mister0wl/arduino-CAN version of the driver and use 1000E3, see if you come right. there is a issue post earlier that talks about this problem.  Issue #62

SomenKumarDas, 2021-06-17:
@altoriot I tried this code as well but it seems not working... Behavior is still the same... I checked what is changed.

This man just added this line 

       if (chip.revision >= 2) {
          modifyRegister(REG_IER, 0x10, 0); // From rev2 used as "divide BRP by 2"
       }

But I checked this IER register is for interrupt control bits, He is basically setting 0 in the Wake-Up Interrupt Enable bit.

mister0wl, 2021-06-20:
"But I checked this IER register is for interrupt control bits, He is basically setting 0 in the Wake-Up Interrupt Enable bit."

Nope. It seems I have to refer once more: https://esp32.com/viewtopic.php?f=12&t=2142&sid=ceb7c54916bd0c1c48dcdf732a0cc26a&start=10
The meaning of this bit has changed for v2. You were looking to the v1 documentation. But let's make it simple. Just try it and if it works, you wil lbe happy, if not, you won't be more sad than now. 
Cheers mate

altoriot, 2021-06-20:
@mister0wl I'm working along with neliusNDL, He incorporated the code changes you suggested back in April. However with my testing, I haven't managed to get the CANBus to operate at 1M. So far for our needs, 500k has been fine, but we will need 1M at some point. When I get a chance ill look deeper into it as I need to get an aspect of the SJA1000 chip working which this driver doesnt provide. (CANBUS has annoying habit of loosing data on the fifo buffer and then goes out of sync with garbage data in the packet structure... bla bla) Ill revert back here when I've poked at this with a big stick.

SomenKumarDas, 2021-06-22:
Yeah @mister0wl you are right about the IER bit, but still I'm facing the issue that 1mbps is not working... after setting wakeup bit  to 0, brp is not dividing also 500kbps is working in 500kbps but when i set can to 1mbps, Can bus showing busoff....

Problem is still the same...

RudolfAtHome, 2021-07-03:
I also have problems with 2 different ESP32 variants, both of which do not work with the CAN library. My ESP32 runs in a CAN bus with 2 Arduino Nano, one sends data, the other (and the ESP32) should only receive data. The target speed is 500 kB / s (500E3).
To determine the chip version I have the command sequence
```
   esp_chip_info_t chip;
   esp_chip_info (& chip);

```copied into my sketch, the 2nd line of the declaration is displayed as an error. After commenting out the second line, the chip version = 0 is displayed.
So the correction doesn't do anything for me.
I also started with 1000E3 once - no success!
The very "stable" result of the ESP32 in loop () with S = packetSize:
S = 0, REG_SR = 0x3C
S = 0, REG_ECC = 0x00

RudolfAtHome, 2021-07-03:
Additional: sorry - I placed the chip-info-lines at a wrong place. At the correct place I get chip version = 1. But it does not help...

EmbeddedEric, 2022-03-09:
Hi,

after setting up the baudrate, the function calls:
`writeRegister(REG_IER, 0xff); // enable all interrupts`

so the fix for half baudrate is overwritten by writing the full register with 0xFF. With that write you activate the divider of 2!

Solution is to use modifyRegister to activate needed interrupts with correct mask or clear the Bit after writeRegister(REG_IER, 0xff); like that:

```
writeRegister(REG_IER, 0xff); // enable all interrupts

if (chip.revision >= 2) 
{
    modifyRegister(REG_IER, 0x10, 0); // From rev2 used as "divide BRP by 2"
}
```