sandeepmistry/arduino-CAN #77: [FR] Example file for filter

italocjs, 2021-10-01:
A example for how to use the filter would be really appreciated, i cant get it working :(

timurrrr, 2021-10-01:
MCP2515 or ESP32?

If MCP2515, see https://github.com/sandeepmistry/arduino-CAN/pull/47 where I fixed a couple of things that might be getting in your way.

italocjs, 2021-10-01:
Its ESP32 + a MPC2515 as transceiver,  the issue i have is that if i read the CAN every loop  without delay, it works fine, but if i only read it when its needed (~10ms) it always return the same value, its like a buffer or something is getting stuck.

I have setup a can testing using the pcan program (with a peak transceiver), sending several uneeded data (simulating what i will get when i connect to a vehicle).

I tried the filtering, and i'm not sure if its working, but the issue is the same as before implementing it.

the only ID i care about is 0x12, so i used CAN.filter(0x12, 0x1FFFFFFF);  in setup.

italocjs, 2021-10-01:
I'm closing this, as i have managed to make it work and added a PR for a new example.
#78

closed 2021-10-01 by italocjs