sandeepmistry/arduino-CAN #84: Starting CAN failed!

mianqi2016, 2022-01-06:
I used a module with MCP2515 + TJA1050 on Arduino UNO, I tried 2 modules on 2 UNO with sender and receiver example, and I have changed parameters for MCP2515, but it just prompted: Starting CAN failed!

  CAN.setSPIFrequency(4E6);
  CAN.setClockFrequency(8E6);
  if (!CAN.begin(10E3)) 

What maybe the reason?

qubolino, 2022-01-09:
does it start without changing spi and clock freq?

mianqi2016, 2022-01-13:
No, so I went to change them.

My wiring was here: https://forum.arduino.cc/t/starting-can-failed/944127/4

mianqi2016, 2022-01-19:
> does it start without changing spi and clock freq?

No.

jackievlobos1, 2022-02-08:
I am having the same issue. Did you find a solution?

mianqi2016, 2022-02-19:
I found if you select "ACAN" in Arduino IDE "Library Manager", you could meet others CAN libraries, I havn't try them, maybe there is any could run.