sandeepmistry/arduino-CAN #86: Feature Request: ESP32 calls the ESP-IDF TWAI driver implementation

Dazza0, 2022-02-21:
Hi @sandeepmistry,

I was just wondering if any thought has been given to updating the ESP32 support to call the [ESP-IDF TWAI driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html) API instead. Calling the ESP-IDF driver will allow the ESP32 Arduino users to get support for the new HW features (in the ESP32-S2) and the HW errata workarounds. We can create a PR for this update if that is OK with you.

pierreSCpapon, 2022-04-14:
This would also be useful for the ESP32-C3 which also uses the TWAI driver.

johnrbnsn, 2022-06-10:
@Dazza0 - If you have code for a pull request that would work, I'd love to be able to use it. I'm not affiliated with arduino-CAN, but we're trying to use an ESP32-S3 with CAN, and would love to stay in Arduino land. Can you post a repo? 

Thanks!

Sheimy, 2022-10-25:
this is the idf TWAI implantation
Try This one i try it with S3
https://github.com/Sheimy/TWAI_S3_LISEN

Subsea765, 2023-04-09:
@Sheimy How do I use the TWAI library with the Arduino IDE? I have tried to use a project by Morcibacsi called PSAWIFI but it does not work on the current ESP32 Devkit V4. Nothing appears at the CanTX pin. I think the board is not compatible with the older CAN library by Sandeep Mistry.