sandeepmistry/arduino-CAN #91: Support for MCP2515 with 12MHz oscillator

carsondarling, 2022-04-25:
I was recently attempting to use a  CAN board designed for a Raspi ([Waveshare's CAN Hat](https://www.waveshare.com/rs485-can-hat.htm)) with this library, but it has a 12MHz crystal attached to the MCP2515. I was able to modify MCP2515.cpp to support the 12MHz oscillator for my use case below. Is there interest in me putting together a pull request for full 12MHz support in this library?

```diff
diff --git a/src/MCP2515.cpp b/src/MCP2515.cpp
index aa46051..e6afd08 100644
--- a/src/MCP2515.cpp
+++ b/src/MCP2515.cpp
@@ -103,6 +103,8 @@ int MCP2515Class::begin(long baudRate)
     {  (long)8E6,   (long)10E3, { 0x0f, 0xbf, 0x07 } },
     {  (long)8E6,    (long)5E3, { 0x1f, 0xbf, 0x07 } },
 
+    { (long)12E6,  (long)500E3, { 0x00, 0xa2, 0x02 } },
+
     { (long)16E6, (long)1000E3, { 0x00, 0xd0, 0x82 } },
     { (long)16E6,  (long)500E3, { 0x00, 0xf0, 0x86 } },
     { (long)16E6,  (long)250E3, { 0x41, 0xf1, 0x85 } },
```

hellmutbrudny, 2022-07-16:
Hi. I found this one looking for a solution for the same problem. But... I wanted to run my CAN HAT with 250kbps (this is a standard in my network). How did you calculated your values for 500kbps?

carsondarling, 2022-07-16:
I used the [datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/MCP2515-Stand-Alone-CAN-Controller-with-SPI-20001801J.pdf)'s description of the CNF registers (section starts on page 39). I know Kvaser has a [calculator ](https://www.kvaser.com/support/calculators/bit-timing-calculator/), though I haven't used it.

carsondarling, 2022-07-16:
Looking at the Kvaser calculator, it seems like you could use something like this:

```diff
+    { (long)12E6,  (long)250E3, { 0x01, 0x9b, 0x02 } },
```
(fair warning: this is totally untested, YMMV, etc)

hellmutbrudny, 2022-07-16:
Your values works. Thanks A LOT!

magicsmoke, 2023-06-24:
Yeah, this could defo be added to the library.

carsondarling, 2023-06-26:
For what it's worth, I've since started using [timurrrr/arduino-CAN](https://github.com/timurrrr/arduino-CAN) (a fork of  this library) since it appears that @sandeepmistry isn't updating this version.