sandeepmistry/arduino-CAN #96: Communication issues with SocketCAN

paolotherocker, 2022-08-15:
I am seeing some strange behaviour when trying to send frames to a Linux machine with [SocketCAN](https://www.kernel.org/doc/html/latest/networking/can.html).

This is the setup:
Arduino MKR Zero with Arduino MKR CAN Shield (terminating resistor active) -> Kvaser USBcan -> Laptop with Ubuntu 22

this is the sketch:
```
#include <CAN.h>

uint32_t g_time = 0;
uint32_t g_led_time = 0;

void setup()
{
    CAN.setSPIFrequency(1E6);

    // start the CAN bus at 500 kbps
    while (!CAN.begin(500E3))
    {
    }

    pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
    const uint32_t current_time = millis();

    if (current_time - g_time >= 500)
    {
        g_time = current_time;

        bool success = false;

        CAN.beginExtendedPacket(419412872);

        CAN.write(0x11);
        CAN.write(0x22);
        CAN.write(0x33);
        CAN.write(0x44);
        CAN.write(0x55);
        CAN.write(0x66);
        CAN.write(0x77);
        CAN.write(0x88);

        CAN.endPacket();

        CAN.beginPacket(0x3F);

        CAN.write(0x11);
        CAN.write(0x22);
        CAN.write(0x33);
        CAN.write(0x44);
        CAN.write(0x55);
        CAN.write(0x66);
        CAN.write(0x77);
        CAN.write(0x88);

        CAN.endPacket();

        digitalWrite(LED_BUILTIN, HIGH);
        g_led_time = current_time;
    }
    if (current_time - g_led_time >= 10)
    {
        digitalWrite(LED_BUILTIN, LOW);
    }
}
}
```

on Linux:
```
sudo ip link set up can0 type can bitrate 500000
candump can0
```

TLDR: **No frames come thorough on Linux** 
Some more details:
- I can <sometimes> see standard frames, if I only send standard frames, with no extended frames.
- Works OK with a laptop running Windows and [BUS Master](https://rbei-etas.github.io/busmaster/). I can see all of the messages with no error frames.
- I can connect a different CAN sensor to the same setup, and I can see all of the messages on Linux, in fact it is the first time I have any problem with SocketCAN.
- If I use a different CAN to USB device (I tried another Kvaser and a Peak device), the same problem occurs.
- I have tried 3 different arduino MKR boards (one MKR WiFi 1010, and two MKR Zero) and CAN shields, same problem.
- If I connect two CAN->USB dvices to two laptops, one running Windows and one running Linux SocketCAN, I can see all of the messages on the Windows machine and none on the Linux machine, however when the Linux machine is connected I see error frames on the Windows machine, namely "Stuff Errror" (ECC `10101010`), at least once, sometimes periodically.

I would really appreciate any suggestion on diagnosing the problem. None of this indicates it is a hardware issue, and SocketCAN never failed me before, so I am inclined to think I am doing something wrong with the library, or there is a bug somewhere.

Thanks.

emekBaris, 2022-11-01:
Having the exact same problem with MKR WIFI and MKR CAN shield. Have you found a solution?

paolotherocker, 2022-11-01:
Unfortunately not, still having the same issue. I found that other CAN enabled devices can also see the messages no problem, it's just Linux Socket CAN that has trouble

emekBaris, 2022-11-01:
I see, thanks! I believe my issue was related to the fact that CANH and CANL are not written correctly on the shield. The correct order is VIN, CANH, CANL, GND. Now, I can send messages on Arduino and can see them by candump. I use Ubuntu 20.04 and both innomaker and peak x6 can adapters worked.

However, I can not receive messages on Arduino sent by socketcan. I can receive messages sent by other devices. Were you able to receive messages on Arduino sent by socketCAN?

silk-indus, 2023-11-28:
Have you checked the bus frequency? Do they match?