sandeepmistry/arduino-CAN #98: Guru Meditation Error:  using SN65HVD230 on ESP32 with callback

MirkoUgoliniDev, 2022-09-29:
Hello all,

I'm using a ESP32 with external   SN65HVD230 and the code with callback, but I got   this error:

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Someone enconterd the same problem?

regards

Mirko

RSM123456, 2023-02-07:
Yes , I also encountered the same problem while using ESP32 with external SN65HVD230 and code with callback function.
Regards
RSM

ZephyrCloudNine, 2023-03-02:
Hey guys, I had the issue as well. I think the reason being that the sketch runs entirely inside an ISR, and the main loop is empty. ESP32's watchdog timer is trigger happy so since the ISR has so many serial prints it throws a watchdog timer error. Simple fix, have the callback set a boolean flag and then do all the Serial prints inside the main loop. Works for me!