sandeepmistry/arduino-CAN #99: Parsing multiple packages at the same time ESP32

g-philipp, 2022-10-02:
Hey there,

i have an esp32 with can transceiver and i like to listen to multiple packages at the same time. 

My code works like this:

`
void decode() {
  int packetSize = CAN.parsePacket();
  int canData[8];

  if (packetSize) {
    if ((CAN.packetId()) == 0x5C3) {
      for (int i = 0; i < packetSize; i++) {
        canData[i] = CAN.read();
      }
      // do sth with the data via the canData array
    }
    if ((CAN.packetId()) == 0x351) {
      for (int i = 0; i < packetSize; i++) {
        canData[i] = CAN.read();
      }
      // do sth with the data via the canData array
    } else {
    }
  }
}
`

I think the problem is that the packages with the id "5C3" are coming in everytime the if-statement is being executed and the second if (for id "351") isnt getting through

What am i doing wrong?

SidtharthanK, 2023-12-12:
Hi, Did you solve the issue? I am trying to read multiple packets(22) and I am getting only 7 and other packets are duplicates of the 3rd one every time (for all 22 packets).
15:39:35.738 -> CAN Initialized
15:39:35.879 -> Received extended packet with id 0x180100FE and length 8
15:39:35.879 -> 1 34 0 39 0 5c b4 4 
15:39:35.879 -> Received extended packet with id 0x180200FE and length 8
15:39:35.926 -> 2 82 c 83 c 84 c ff 
15:39:35.926 -> Received extended packet with id 0x180300FE and length 8
15:39:35.926 -> 3 85 c 86 c 87 c ff 
15:39:35.926 -> Received extended packet with id 0x180400FE and length 8
15:39:35.926 -> 4 88 c 89 c 8a c ff 
15:39:35.926 -> Received extended packet with id 0x180500FE and length 8
15:39:35.926 -> 5 8b c 8c c 8d c ff 
15:39:35.926 -> Received extended packet with id 0x180600FE and length 8
15:39:35.926 -> 6 8e c 8f c 90 c ff 
15:39:35.926 -> Received extended packet with id 0x180700FE and length 8
15:39:35.926 -> 7 16 17 18 19 1a 1b 88 
15:39:35.926 -> Received extended packet with id 0x180300FE and length 8
15:39:35.926 -> 3 85 c 86 c 87 c ff 
15:39:35.926 -> Received extended packet with id 0x180300FE and length 8
15:39:35.972 -> 3 85 c 86 c 87 c ff 
15:39:35.972 -> Received extended packet with id 0x180300FE and length 8
15:39:35.972 -> 3 85 c 86 c 87 c ff 
15:39:35.972 -> Received extended packet with id 0x180300FE and length 8
15:39:35.972 -> 3 85 c 86 c 87 c ff 
15:39:35.972 -> Received extended packet with id 0x180300FE and length 8
15:39:35.972 -> 3 85 c 86 c 87 c ff 
15:39:35.972 -> Received extended packet with id 0x180300FE and length 8
15:39:35.972 -> 3 85 c 86 c 87 c ff 
15:39:35.972 -> Received extended packet with id 0x180300FE and length 8
15:39:35.972 -> 3 85 c 86 c 87 c ff 
15:39:35.972 -> Received extended packet with id 0x180300FE and length 8
15:39:35.972 -> 3 85 c 86 c 87 c ff 
15:39:36.018 -> Received extended packet with id 0x180300FE and length 8
15:39:36.018 -> 3 85 c 86 c 87 c ff 
15:39:36.018 -> Received extended packet with id 0x180300FE and length 8
15:39:36.018 -> 3 85 c 86 c 87 c ff 
15:39:36.018 -> Received extended packet with id 0x180300FE and length 8
15:39:36.018 -> 3 85 c 86 c 87 c ff 
15:39:36.018 -> Received extended packet with id 0x180300FE and length 8
15:39:36.018 -> 3 85 c 86 c 87 c ff 
15:39:36.018 -> Received extended packet with id 0x180300FE and length 8
15:39:36.018 -> 3 85 c 86 c 87 c ff 
15:39:36.018 -> Received extended packet with id 0x180300FE and length 8
15:39:36.018 -> 3 85 c 86 c 87 c ff 
15:39:36.018 -> Received extended packet with id 0x180300FE and length 8
15:39:36.065 -> 3 85 c 86 c 87 c ff 
15:39:36.065 ->

Paulie92, 2023-12-24:
Hi, I have similar problem. I am sending 60 packets from one ESP32 to another one every 50ms- The first 3 packets from the group can be read normal, but 5th to 60th packet is updated approx. once a second or two. I assume there is maybe some problems with the buffer. Does anyone find a solution for sending and receiving multiple packets?

SidtharthanK, 2024-01-02:
Hi, I increased the timing between each packet transmission to 5 to 10ms Gap so that this issue does not appear.