sandeepmistry/arduino-CAN #106: MCP2515 observe() implementation seems incorrect.

ddejean, 2022-11-16:
https://github.com/sandeepmistry/arduino-CAN/blob/382fff7a04673705fbd5bc79e6badd319a03c3a8/src/MCP2515.cpp#L356

I'm currently looking at the `observe()` method implementation for MCP2515 and I'm wondering if the line referenced above is correct. According to the [datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/MCP2515-Stand-Alone-CAN-Controller-with-SPI-20001801J.pdf), it seems the 0x80 value is for "Configuration mode". If I read the datasheet correctly, the value to use here should be 0x60.

What do you think ?

sandeepmistry, 2022-11-19:
@ddejean would you be able to make a pull request for this change? Thanks.

ddejean, 2022-11-20:
Sure I'm on it :)

closed 2022-12-04 by sandeepmistry