sandeepmistry/arduino-CAN #112: ESP8266 support

Jojo-A, 2023-01-17:
Hi,

although it is not explicitly mentioned to be supported, I tried to get this library running on a ESP8266 board (NodeMCU).
After my first implementation, the compilation failed with:
```
Arduino/libraries/CAN/src/MCP2515.cpp:268:9: error: 'class SPIClass' has no member named 'usingInterrupt'
268 |     SPI.usingInterrupt(digitalPinToInterrupt(_intPin));
      |         ^~~~~~~~~~~~~~

```
After a little bit of research I got to the conclusion, that the interrupt structure of the ESP8266 might just not (yet) be supported.
So I went into the MCP2515.cpp and the MCP2515.h and removed all references to anything that had something to do with interrupts.
in the .h file:
```
//  virtual void onReceive(void(*callback)(int));
//  void handleInterrupt();
//  static void onInterrupt();
```
and of course the corresponding functions in the .cpp file.
And voilà, compiled successfully and runs fine on my NodeMCU board (tested with a PCAN-USB dongle).

So it would be nice to see the ESP8266 support to get it working with interrupts OOTB. It would make this library even more versatile and would target many more people out there!

Greetings

Jojo-A, 2023-01-18:
Okay, I got a just closer. Instead of commenting out all stuff about interrupts, I did just
```
  if (callback) {
    #ifndef ARDUINO_ARCH_ESP8266
    SPI.usingInterrupt(digitalPinToInterrupt(_intPin));
    #endif
```
in the MCP2515.cpp file. That compiles successful. Next step is that I will test it if it works on the ESP8266 _with_ interrupts.

timurrrr, 2023-01-26:
Curious why you need ESP8266 support.
It's a pretty old chip with known security limitations, and there's already a superior ESP32-C3 on the market.

Jojo-A, 2023-01-30:
> 

You are totally right about that!
But I would not argument with my personal needs or requirements.
The main argument would be, that many people are *currently* using the ESP8266. So supporting that one could be a benefit for all those existing people out there. Many people (like me) also just have some ESP8266 boards on stock. So during this time of transition to the newer chips the support for the "older" ones would be something that many people benefit from.
I agree that future projects should more go for the newer chips.

But I also think that people should switch to your fork of this library because of known limitations of that one here...