sandeepmistry/arduino-CAN #115: When compiling a file for esp32 s3 dev board, i get the following errors. Is there a new CAN.c file to help?

Phil-Mitton, 2023-01-30:
In file included from C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_reg.h:21,
                 from c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:39:
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c: In function 'CAN_init':
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:174:26: error: 'DPORT_PERIP_CLK_EN_REG' undeclared (first use in this function); did you mean 'SYSTEM_PERIP_CLK_EN1_REG'?
  DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                          ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:174:2: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
  DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
  ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:174:26: note: each undeclared identifier is reported only once for each function it appears in
  DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                          ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:174:2: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
  DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
  ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:174:50: error: 'DPORT_CAN_CLK_EN' undeclared (first use in this function); did you mean 'DPORT_REG_CLR_BIT'?
  DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                                                  ^~~~~~~~~~~~~~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:88:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:174:2: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
  DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
  ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:176:28: error: 'DPORT_PERIP_RST_EN_REG' undeclared (first use in this function); did you mean 'SYSTEM_PERIP_RST_EN0_REG'?
  DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
                            ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                   ^~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:85:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
                                              ^~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:176:2: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
  DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
  ^~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:176:52: error: 'DPORT_CAN_RST' undeclared (first use in this function); did you mean 'DPORT_DATE_REG'?
  DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
                                                    ^~~~~~~~~~~~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:75:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                                        ^~~
C:\Users\Phil Mitton\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_access.h:85:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
                                              ^~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:176:2: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
  DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
  ^~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:181:37: error: 'CAN_TX_IDX' undeclared (first use in this function); did you mean 'TWAI_TX_IDX'?
  gpio_matrix_out(CAN_cfg.tx_pin_id, CAN_TX_IDX, 0, 0);
                                     ^~~~~~~~~~
                                     TWAI_TX_IDX
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:186:36: error: 'CAN_RX_IDX' undeclared (first use in this function); did you mean 'TWAI_RX_IDX'?
  gpio_matrix_in(CAN_cfg.rx_pin_id, CAN_RX_IDX, 0);
                                    ^~~~~~~~~~
                                    TWAI_RX_IDX
c:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster\src\CAN.c:256:17: error: 'ETS_CAN_INTR_SOURCE' undeclared (first use in this function)
  esp_intr_alloc(ETS_CAN_INTR_SOURCE, 0, CAN_isr, NULL, NULL);
                 ^~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "ESP32CAN.h"
  Used: C:\Users\Phil Mitton\Documents\Arduino\libraries\ESP32ArduinoCANmaster
  Not used: C:\Users\Phil Mitton\Documents\Arduino\libraries\esp32-can-protocol-demo-master
exit status 1

Compilation error: exit status 1

mattia424, 2023-04-18:
I have the same problem. Is there a solution for this?
Thank you

JeseRamanantsoa, 2023-05-08:
Any solution guys?

ruziev-dev, 2023-05-24:
The same shit. Does anybody found solution? 🥺

Loongle, 2023-08-16:
Any solution guys?

dmn8s5, 2024-04-29:
What's going on here? Any update here?

Phil-Mitton, 2024-04-30:
All good, Thank you.
Phil

dmn8s5, 2024-04-30:
Hey Phil, how did you resolve the error? Is there a new update I am not aware of?

bionicbone, 2024-07-13:
@Phil-Mitton @dmn8s5 @per1234 @Loongle @ruziev-dev @JeseRamanantsoa @mattia424 
Did anyone find an alternative CAN driver that will compile on the S3, my particular application is to use 2 * MCP2515, I have this working on an original ESP32 but want to leverage the low power deep sleep of the S3. Thanks all

ZillKhan, 2025-03-21:
Hello,
Please check this fix: https://github.com/sandeepmistry/arduino-CAN/pull/155