sandeepmistry/arduino-CAN #116: Is this lib dead?

Jojo-A, 2023-01-31:
Hi,

I see that this lib has a lot of really great PRs pending which include serious bug fixes and a lot of performance improvements. Especially the PRs from @timurrrr.
Many issues are unanswered as well as many topic in the discussions.
The last commits cover minor changes (link fix... WHO CARES?!?), no real improvements.

So: is this lib dead?!? Maybe it should be removed from the Arduino list or replaced by a better maintained fork?

thalesmaoa, 2023-05-29:
Hi @Jojo-A. Did you found a maintained alternative?

thalesmaoa, 2023-05-30:
Thanks, I will try it. Probably the bug I've noticed is not present there.

zalexzperez, 2023-07-15:
> It runs rock-stable in my application.

Are you using VScode + Platformio?
What platform version? 

After hours of investigation, I found that using a version of platform-espressif32 newer than 3.5.0 (and 4.2.0 in another case) makes the ESP32 reboot in a loop when connected to a CANBUS network (95kbps). Using ESP32 devkitC v1 and v4 along with SN65HVD230 transceiver combo.

A simple sniffing sketch like this makes the device reboot in a loop:
https://github.com/adamtheone/canDrive/tree/main/01_canSniffer_Arduino

Recent versions of the Arduino IDE (2.x) have the same bad outcome. Arduino IDE 1.8.19 (and platform-espressif32 @ 3.5.0) work fine.

Jojo-A, 2023-07-18:
> Are you using VScode + Platformio?
> What platform version?

Oh, this is some time ago already.
I just used Arduino IDE v2.0.3 and the BSP for esp8266 3.1.1 and the BSP for ESP32 2.0.6 (iirc).
Please open another issue to discuss your bootloop problem. I started this issue to find out how good (or bad) this lib is maintained.

elementzonline, 2023-10-12:
This issue helped to solve my problem. I was pulling out my hair for 6 hours with 2 esp32 wroom module connected  with MCP2551 transceiver chip. And I finally came across this issue. 

The fork mentioned worked out of the box. 
https://github.com/timurrrr/arduino-CAN

Anyway thanks for this wonderful Library.