sandeepmistry/arduino-CAN #123: ESP32 - only 1 extended filter working, is it possible to specify multiple filters?

rosak, 2023-10-12:
First of, thank you very much for that useful library.

Now:

    CAN.filterExtended(0x04394000);
    CAN.filterExtended(0x06214000);

No matter how many filters I set it only reads the last one, is it the library limitation or am I doing something wrong?

Thanks in advance.

silk-indus, 2023-11-23:
I added the dual filter.

rosak, 2023-11-24:
Thank you very much @silk-indus, really appreciated. I'll see what changes you made and try to add another one myself as I need at least 3.

silk-indus, 2023-11-28:
I have used the hardware support of the filtering (I only need to check hardware registers), i.e. no software check is required. The packets are filtered on the hardware level and as far as I know, data are even not stored in the module. However, I think it is possible to use more than two filters, but another software check is required. This could be done in the onReceive call.

Davi512, 2024-01-23:
> I added the dual filter.

Where?

silk-indus, 2024-01-23:
> > I added the dual filter.
> 
> Where?

Please look on my fork:
https://github.com/silk-indus/CAN-ESP32

Davi512, 2024-01-23:
You mean this ?

"virtual int filters(int id1, int mask1, int id2, int mask2);" 

SO I can use filter for 2 ID's?

What about 3 or more? Some mask?

silk-indus, 2024-01-23:
Yes, exactly.
Two filters are supported by the hardware, i.e., this is evaluated on hardware level, so you don't need to do anything more. If you want to have sophisticated filtering, you must do it in software way, for instance check on every message if ID match the conditions. This solution doesn't reduce the traffic and moreover reduce the computation power.

Davi512, 2024-01-23:
Ok, what about filtering by mask? 
Maybe this way is better for more ID's?

silk-indus, 2024-01-23:
OK, mask is like a pattern. The HW way is to write the pattern to the HW register, and the HW do the filtering job. If HW is unable to process more than 2 filters (doesn't have more filter registers, what is our case), you have no other ways, just to filter the packets by the SW. You can use the same patterns, but all the stuff is up to you. That means you have to check every incoming packet (OR, XOR, … whatever, depending on your imagination) if matching your criteria.

Davi512, 2024-01-24:
I use filter because I have a lot of ID's and I lose some of them. 
I tried oncallback but the cpu is restart without filter

silk-indus, 2024-01-25:
I think, you have a different problem. If the bus is overloaded, filtering does not solve your problem. Filtering can reduce the job of the receiving device. If the bus is ok, then most probably you are not properly receiving the packets. The good practice is to receive packets using interrupts (not by polling).

Davi512, 2024-01-25:
I use callback and write data to LCD and is ok. But when I run sample callback receive and uart my cpu restart after moment [image]

silk-indus, 2024-02-05:
Sorry for the delay.
This error is likely connected with insufficient time for the interrupt. In other words, the next interrupt comes before the old one finishes. A general rule says to reduce interrupt instructions. A common mistake is Serial.print() instruction in the interrupt. Please definitely avoid it.