sandeepmistry/arduino-CAN #129: Working with MCP2515 on Arduino Nano, but fails on ESP32

cgavaller2, 2024-01-28:
I have tried everything before resorting to reporting my issue here.

I have an 8Mhz MCP2515 board wired to an ESP32 dev board (Already configured the correct frequency in the MCP2515.h) that just simply won't work. None of the built-in examples, nothing. I have tried the SPI pins for both VSPI and HSPI to no avail. I have also tried reassigning the SPI pins and the functionality is the same. Under the built-in examples, when the library attempts to establish a connection, it prints a success message to the serial monitor, even with the MCP board disconnected from the ESP??? I have added print statements to the VIN Reader code and found that the example hangs at the CAN.endPacket(); function. Although my code and all examples work fine and can connect to my vehicle when testing on an Arduino Nano???? I can use the nano as a last resort, but I'd much rather use the ESP.

Any Ideas?

cgavaller2, 2024-01-29:
I believe the issue is that the library recognizes it's on the ESP32 architecture and defaults to the inbuilt CAN controller. How can I force the MCP to be the one used?

zalexzperez, 2024-01-30:
That MCP2515 is a module that comes with a TJA1050 controller, right? The CAN controller works at 5V logic levels, not 3.3V.
Get a genuine SN65HVD230 module (there are many with fake chips out there). It'll work well with this library if you're using the old ESP32.
You can also try using the native TWAI driver.