sandeepmistry/arduino-CAN #138: #warning esp_intr.h is deprecated

atestani, 2024-06-14:
I am using the library in an ESP32 application in PlatformIO using the Arduino framework with MCP2515 hardware. I am getting the warning message below.  It is only a warning, so no big deal, but I thought you would like to know.

```
In file included from .pio/libdeps/esp32dev/CAN/src/ESP32SJA1000.cpp:6:
C:/Users/Al Testani/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
```

zetazetterberg, 2024-07-04:
replaced with esp_intr_alloc.h.

ZillKhan, 2025-03-21:
Hello,
Please check this fix: https://github.com/sandeepmistry/arduino-CAN/pull/155