sandeepmistry/arduino-CAN #139: setPins() for rx/tx and cs/irq?

atestani, 2024-06-14:
I am using an MCP2515 on an ESP32 and writing code in PlatformIO/VSCode with the Arduino framework.   I need to set both rx/tx and cs/irq  from the defaults.  I see in the API.md that there are two calls to setPins() for both pairs of pins.  How does the library know which set of pins I am defining?

alastaira, 2024-10-02:
That depends on whether you're targetting ESP32 or AVR architecture.

brightproject, 2024-10-20:
I also didn't quite understand the function:
```
void MCP2515Class::setPins(int cs, int irq)
{
_csPin = cs;
_intPin = irq;
}
```

> https://github.com/sandeepmistry/arduino-CAN/blob/b514e6b8596126a8f94f226b9906455393efd3a2/src/MCP2515.cpp#L396

According to the code `CAN.setPins(PA4, -1);` there is an explicit installation of the pin of the `CS` and `INT` and nothing else, the platform has nothing to do with it.

alastaira, 2024-11-09:
> I also didn't quite understand the function:
>
> ```
> void MCP2515Class::setPins(int cs, int irq)
> {
> _csPin = cs;
> _intPin = irq;
> }
> ```
>
>
> According to the code `CAN.setPins(PA4, -1);` there is an explicit [...]

Look at the top of the header file... the MCP2515 class can't be used on ESP32: https://github.com/sandeepmistry/arduino-CAN/blob/b514e6b8596126a8f94f226b9906455393efd3a2/src/MCP2515.h#L4

And the SJA1000 can _only_ be used on ESP32 hardware: https://github.com/sandeepmistry/arduino-CAN/blob/b514e6b8596126a8f94f226b9906455393efd3a2/src/ESP32SJA1000.h#L4

In other words, platform has everything to do with it ;)