sandeepmistry/arduino-CAN #143: Error compiling with Adafruit Feather v2: `esp_intr.h: No such file or directory`

midgarth, 2024-07-22:
When trying to use Examples on an Adafruit ESP32 Feather V2  with tag `0.3.1` getting following error:
```
/Arduino/libraries/CAN/src/ESP32SJA1000.cpp:6:10: fatal error: esp_intr.h: No such file or directory
    6 | #include "esp_intr.h"
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1
```
Attempted to update the ESP32SJA1000.cpp to use `esp_intr_alloc.h. `as mentioned this issue : https://github.com/sandeepmistry/arduino-CAN/issues/138

When doing so get the following errors:
```
Board Manager URLs included: https://dl.espressif.com/dl/package_esp32_index.json,https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
```

I apologize if this is a very basic error/question.

nickskethisnikske, 2024-07-25:
Go to boards manager and search for ESP32, then go back to an older version, 3.0.3 gave the same problems you've mentioned.
Going back to for example 2.0.11 will solve this.

ZillKhan, 2025-03-21:
Hello,
Kindly try with these changes: #155 
These changes might fix the issue.