sandeepmistry/arduino-CAN #146: Solved: Fatal Error: esp_intr.h: No such file or directory

onlyvinod56, 2024-08-23:
Use ESP32 board ver 2.0.11 in board manager. 
(Espressif Systems)

[image]

bluecraank, 2024-09-10:
Same, help pls

Merange, 2024-09-25:
It seems to be a bug in V3
I came back to the latest V2 V2.0.17 and it works fine

avlasic, 2024-09-27:
#150 pull request should fix this problem in v3

JayMcAnser, 2024-10-26:
This article describes the file has been remove:
https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/migration-guides/release-5.x/system.html

drnet-at, 2024-11-10:
Replace 

//#include "esp_intr.h"

with

#include "rom/gpio.h"

In src/ESP32SJA1000.cpp to get it running

BWV999-MTL, 2025-01-23:
Hi all, i tried the CANSender.ino sketch. my setup is esp32-S3 dev module. the fix mentionned above compiles for a DOIT ESP32 DEVKIT V1, but i get the following errors for esp32-S3 dev module :

In file included from C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_reg.h:21,
                 from C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:8:
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp: In member function 'virtual int ESP32SJA1000Class::begin(long int)':
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:57:29: error: 'DPORT_PERIP_RST_EN_REG' was not declared in this scope; did you mean 'SYSTEM_PERIP_RST_EN1_REG'?
   57 |   DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
      |                             ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:77:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
   77 | #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
      |                                                                   ^~~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:87:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
   87 | #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
      |                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:57:3: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
   57 |   DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:57:53: error: 'DPORT_CAN_RST' was not declared in this scope
   57 |   DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
      |                                                     ^~~~~~~~~~~~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:77:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
   77 | #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
      |                                                                                        ^~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:87:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
   87 | #define DPORT_CLEAR_PERI_REG_MASK(reg, mask) DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)&(~(mask))))
      |                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:57:3: note: in expansion of macro 'DPORT_CLEAR_PERI_REG_MASK'
   57 |   DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:58:27: error: 'DPORT_PERIP_CLK_EN_REG' was not declared in this scope; did you mean 'SYSTEM_PERIP_CLK_EN1_REG'?
   58 |   DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
      |                           ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:77:67: note: in definition of macro '_DPORT_WRITE_PERI_REG'
   77 | #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
      |                                                                   ^~~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:90:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
   90 | #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
      |                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:58:3: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
   58 |   DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:58:51: error: 'DPORT_CAN_CLK_EN' was not declared in this scope
   58 |   DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
      |                                                   ^~~~~~~~~~~~~~~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:77:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
   77 | #define _DPORT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
      |                                                                                        ^~~
C:\Users\Public\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/soc/esp32s3/include/soc/dport_access.h:90:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
[... cut ...]
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp: In member function 'virtual void ESP32SJA1000Class::onReceive(void (*)(int))':
C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:284:20: error: 'ETS_CAN_INTR_SOURCE' was not declared in this scope
  284 |     esp_intr_alloc(ETS_CAN_INTR_SOURCE, 0, ESP32SJA1000Class::onInterrupt, this, &_intrHandle);
      |                    ^~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

I replicated the issue on several computers and fresh installs of Arduino IDE 2.3.2. Any ideas?

drnet-at, 2025-01-23:
The line

C:\Users\Public\Arduino15\libraries\CAN\src\ESP32SJA1000.cpp:57:29: error: 'DPORT_PERIP_RST_EN_REG' was not declared in this scope; did you mean 'SYSTEM_PERIP_RST_EN1_REG'?

Tells you everything you need to know: The macro DPORT_PERIP_RST_EN_REG is obviously not resolved in your version of the ESP32 for Arduino SDK. You need to either track it down (if it is now in another header file) and include it, or replace it by the macro which is valid for your SDK (if it was renamed).

BWV999-MTL, 2025-01-24:
I had a look a at the technical reference manual for both esp32 and esp32 s3.  I might be able to track down registers and migrate them in the library, but i could not find confirmation that the twai controller in the s3 version is still compatible with the SJA1000 as it was in esp 32. I might be in over my head with this. My end goal is to use the Odrive arduino library with an S3 and builtin TWAI (https://github.com/odriverobotics/ODriveArduino). instead of trying to migrate sandeepmistry's arduino-CAN library to esp32 S3, i might have a better chance of succes in implementing a IS_ESP32-S3_BUILTIN Board-specific includes that uses TWAI in Odrives library.