sandeepmistry/arduino-CAN #154: Compatibility with Arduino 3.1.0 - change of internal GPIO api.

mazgch, 2025-01-02:
The library no longer compiles using Arduino_esp32 v 3.1.0 probably internal api names have changed but not sure. 

.../libraries/CAN/src/ESP32SJA1000.cpp: In member function 'virtual int ESP32SJA1000Class::begin(long int)':
.../libraries/CAN/src/ESP32SJA1000.cpp:61:3: error: 'gpio_matrix_in' was not declared in this scope
   61 |   gpio_matrix_in(_rxPin, CAN_RX_IDX, 0);
      |   ^~~~~~~~~~~~~~
.../libraries/CAN/src/ESP32SJA1000.cpp:62:3: error: 'gpio_pad_select_gpio' was not declared in this scope; did you mean 'esp_rom_gpio_pad_select_gpio'?
   62 |   gpio_pad_select_gpio(_rxPin);
      |   ^~~~~~~~~~~~~~~~~~~~
      |   esp_rom_gpio_pad_select_gpio
.../libraries/CAN/src/ESP32SJA1000.cpp:66:3: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
   66 |   gpio_matrix_out(_txPin, CAN_TX_IDX, 0, 0);
      |   ^~~~~~~~~~~~~~~
      |   gpio_iomux_out

minashinne, 2025-02-20:
https://github.com/sandeepmistry/arduino-CAN/issues/146#issuecomment-2466734908

minashinne, 2025-02-20:
Replace
#include "esp_intr.h"
with
#include "rom/gpio.h"
In src/ESP32SJA1000.cpp to get it running

ZillKhan, 2025-03-21:
Hello,
Please check this fix: #155